mavlink/c_uart_interface_example #12: Error reading messages in serial_port.cpp

tlillies, 2015-03-25:
I apologize if this is not the correct place to ask this but I was not sure where to ask. I am using the c_uart_interface_example as a starting point for sending and receiving mavlink for my own project. I am running SITL and then piping that output into a pseudoterminal (so /dev/pts/) and then opening that as a serial port so I can test the interface without the actual hardware. Using mavproxy works fine for receiving data over the pseudoterminal but I am having trouble with the example serial port interface in this code. I get a lot of the messages, but it seems like I am dropping a good amount of them as well. I am not sure why this would be as I was under the impression that pseudoterminals were handled like other serial ports. I tried playing around with everything I could think of and nothing seems to make a difference. Attached is a snippet of the output with debug mode turned on. Thanks for your help!

[image]

tlillies, 2015-03-25:
It is most likely my ignorance rather than something wrong but I just was not sure where to go from here.

aerialhedgehog, 2015-03-30:
Hey Thomas,

I haven't heard of anyone trying this with SITL yet.  You may be the first!  Are you not able to send messages at all?  In my experience dropped packets are normal, mavlink is built to handle these in fact.  However, if you're running the simulation faster than real time, I suspect that you could run into a few issues with baudrates not being aligned.  Great question, though SITL is a bit outside of my ballpark at the moment.

Best
- Trent

tlillies, 2015-03-31:
Trent,

Thanks for the response. I am able to get a lot of messages, in fact I am not sure which messages I am dropping. I have come to the conclusion that the simulation is not the problem because when I do SITL-->/dev/pts/-->mavproxy no packets are dropped at all while if I do SITL-->/de/pts/-->(program using serial_port.cpp to read and then write messages)-->different /dev/pts/-->mavproxy mavproxy reports that I am dropping more messages than not. It could be I am doing something to make the messages out of order because I think that may be how mavproxy detects dropped messages. I have some other things I want to try as well but won't have time until later this week. If you have any other ideas let me know. Thanks for your help.

Tom

tlillies, 2015-03-31:
I think I am going to close this issue because I think it is my own issue rather than an issue with serial_port.cpp.

closed 2015-03-31 by tlillies