mavlink/c_uart_interface_example #14: Dropped packet issue

jhlink, 2015-05-20:
For some reason, after compiling the program and executing the program, no response is received after finding a response and discovering the system ID and component ID. 

I changed debug to true and found that there is a dropped packet. I've been looking through the msg IDs that are posted to the command console, it seems that all msg IDs are handled at some point. A few msgs by some random basis seem to be dropped completely. 

Is this a bug that hasn't been resolved yet?

jhlink, 2015-05-20:
Also, @tilllles seemed to have brought this issue up when he was testing the application using SITL, but I believe the problem extends to actual hardware. 
I'm using a PX4 flight controller.

jhlink, 2015-05-20:
So after debugging for a bit, I've found that a bunch of messages aren't actually being handled. 
Especially three of messages required, which are local_position_ned, position_target_global_int, and highres_imu.

nopeppermint, 2015-05-25:
=> update submodule c_library would maybe be the solution

aerialhedgehog, 2015-05-25:
try checking out these links below, there is an important step regarding
setting permissions for the tty port.  as you mentioned, this example is
only working for PX4 Firmware.

https://groups.google.com/d/msg/px4users/MfzpqObhraY/LJ8o6GZGXWwJ

https://groups.google.com/d/msg/px4users/MfzpqObhraY/QjYtceJmDFkJ

specifically mugesh's post --
1. Install Qgroundcontrol in Ubuntu. This should not take much time.
2. After this is done, we need to set the permission on the USB port
to flash the firmware. Even though the device might appear, you cannot
flash firmware through USB without setting permission right and
turning off the modem manager. This is the tricky part which everyone
will overlook. This can be done as follows:

i.  sudo chmod 666 /dev/ttyACM0 –> sets the proper permission
ii. sudo usermod -a -G dialout name –> replace name with your account
and don't forget to logout/login. This adds your user to the
dialout-group
iii. sudo apt-get remove modemmanager –> modemmanager tends to occupy
a device only briefly, but it's initial tests make trouble for any
firmware upgrade software that reboots the pixhawk.

3. After this is done, you are almost there. Flash firmware to pixhawk
from QGC over the USB.
4. Now screen into pixhawk and press Enter, you will see nsh prompt.
Start the mavlink application on Telem 2 port which is /dev/ttyS2.
5. Now you execute the c_uart example, you will get the output within no
time.

jhlink, 2015-05-25:
So, reflashing the FC did it. 
Just make sure before reflashing, you erase sd card to "clean it" for the new firmware. 

I didn't see successful results until I cleaned the SD card, even after trying the method stated above. 

Closing Issue!! :D

closed 2015-05-25 by jhlink

aerialhedgehog, 2015-05-25:
fantastic!