mavlink/c_uart_interface_example #16: Can't pass GOT VEHICLE SYSTEM ID: 1 GOT AUTOPILOT COMPONENT ID: 1

cmelkor, 2015-08-19:
Hello,

I was trying to use c_uart, but it fails to pass autopilot component id:1. I check this issue https://github.com/mavlink/c_uart_interface_example/issues/14 but its solution doesn't help me because I can't do this step: "Now screen into pixhawk and press Enter, you will see nsh prompt. Start the mavlink application on Telem 2 port which is /dev/ttyS2"
I can't log into nsh prompt.
http://imgur.com/a/BJqsb

I plugged my px4 via USB.
Sorry for my english, I don't speak english too much.
Regards

ZeyadTamimi, 2016-03-12:
Same issue here, i believe its because the pixhawk is never sending the local position NED message, which is required for the code to progress past initializing the read thread.

MHageH, 2016-05-21:
As ZeyadTamimi said, the pixhawk isn't sending the LOCAL_POSITION_NED message.
One way to make this happen is to properly configure the pixhawk with qgroundcontrol software.
Usually after configuration, it will start streaming the required MAVLink messages

jmcwil, 2016-07-18:
I am having a similar issue, but my set up is different.  I have two FTDI cables connected to Serial 4/5 and my host machine.  Both FTDI cables are connected to ground.  I am currently powering pixhawk through the servo bus using an ESC to regulate the power to 5V.

**Details of my setup are as follows:**

I am connected to the Pixhawk NuttX Shell on the serial 5 Rx and Tx pins.  The NuttX Shell is working for me, and I am able to start mavlink on Serial 4, which is /dev/ttyS6 from NuttX.  Running mavlink status in the nsh gives

```
NuttShell (NSH)
nsh> mavlink start -d /dev/ttyS6
INFO  [mavlink] mode: Normal, data rate: 2880 B/s on /dev/ttyS6 @ 57600B
WARN  [mavlink] hardware flow control not supported
nsh> mavlink status

instance #0:
        mavlink chan: #0
        no telem status.
        flow control:   OFF
        rates:
        tx: 0.208 kB/s
        txerr: 0.000 kB/s
        rx: 0.000 kB/s
        rate mult: 1.000
        accepting commands: YES
```

I have a second FTDI cable connecting the Tx and Rx pins of serial 4 to my host machine.  I am running this example with the command ``./mavlink_control -d /dev/ttyUSB1``.  My NuttX shell is still open on /dev/ttyUSB0, my other FTDI connection to Serial 5.  When I run the command, I get 

```
Connected to /dev/ttyUSB1 with 57600 baud, 8 data bits, no parity, 1 stop bit (8N1)

START READ THREAD 

CHECK FOR MESSAGES
Found

GOT VEHICLE SYSTEM ID: 1
GOT AUTOPILOT COMPONENT ID: 1
```
as my output, but nothing else.  I have been able to achieve similar results on Telem2, but I prefer this setup since it seems that I can access both the nsh and the mavlink connection.

**My goal** is to be able to send custom mavlink commands to the pixhawk from my host computer over a serial connection.  I am currently having trouble using qgroundcontrol for configuration, so the above work-around does not seem like it will help me.  Is there another way to fix this from the terminal? I am able to modify and upload firmware to the pixhawk over USB without using QGC.

Any help is appreciated, thank you.

MHageH, 2016-07-19:
@jmcwil What troubles you are facing with QGroundcontrol? This software is required to configure all the sensors on pixhawk, which will allow it to send the LOCAL_POSITION_NED required by the C interface (actually it will block in an infinite loop until it receives the first position message).

jmcwil, 2016-07-19:
@MHageH Thank you for the reply, and for explaining why I need QGroundControl. 

I have been having trouble connecting to Pixhawk with QGC.  I am able to flash the firmware to the pixhawk, but then after I do that, one of two things happens.
1) The Main LED is blue, and goes from dim to bright as it does while breathing, but then immediately does out once it gets to full brightness, and stays off.  Then I get the message "Your Vehicle is not responding.  If this continues shutdown QGroundControl, restart the Vehicle letting it boot completely, then start QGroundControl."  This has been going on for several days and I have tried shutting down QGC multiple times.
2)The main LED breathes once then flashes before stopping. I am connected for about 1 second and then it says "COMMUNICATION LOST" in the top right corner, and a yellow box pops up with "Vehicle did not respond to request for parameters, retrying.

As a note, my Pixhawk is connected to my computer via USB, and it is powered only by the USB during the flash process.  All that I have connected to the Pixhawk is the safety light, and all other ports are empty.

I was able to start the calibration once a few days ago, but then something must have gone wrong, and later that day QGC started giving this output. I am not sure if it is a Pixhawk issue, a firmware issue, or a QGC issue at this point.  Thanks for helping me figure out what is going on!

jmcwil, 2016-07-19:
I had a second Pixhawk that is untouched.  I decided to try the same process with it.  It is able to get to the configuration step, so it must be a problem with my other Pixhawk.  The other Pixhawk's IO Amber LED never comes on, and in QGC, its second green loading bar does not show up.  These things both happen with the second Pixhawk.  Is there a way to fix the first one?

MHageH, 2016-07-19:
@jmcwil I'm afraid it might be a hardware fault, since you can do the configuration fine on the new pixhawk. Try to reflash it twice with QGroundControl, it might be also a flashing problem too.

jmcwil, 2016-07-19:
@MHageH Unfortunately it does seem like a hardware problem with the IO board.  Although I slight variations in LED flashes and QGroundControl output, I am unable to get a stable connection and calibrate the board.  It is possible I accidentally fried it during a test (the board still works outside QGroundControl).  Thanks for helping me figure this out.

On the working pixhawk, is there a way to use pixhawk to send mavlink messages without fully calibrating?  I need to write and test custom mavlink messages, but I currently have no need for an airspeed sensor or a radio, but QGroundControl is requiring me to calibrate them.  I just switched to pixhawk from another piece of hardware and I am using it as an intermediate step in my project, so would prefer to run the firmware without purchasing extra sensors for it if possible.  I would like to be able to send messages over a serial connection to move individual servos.  Could you tell me whether this is possible?  Thanks for being patient with my lack of knowledge.  l am very new to Pixhawk and QGroundControl.  I really appreciate your taking the time to help me!

jmcwil, 2016-07-19:
I figured out how to calibrate without the radio or the airspeed sensor.  For anyone that needs this information, the trick is to change the QGroundControl parameters for COM_RC_IN_MODE and CBRK_AIRSPD_CHK.  Now my only problem is getting Level Horizon calibration to complete.  I keep getting the error message "attitude estimator not running- check system boo."  If you know what causes that message, please let me know.  Thanks!!

MHageH, 2016-07-20:
@jmcwil Tweaking the 2 parameters was the correct step needed. As for the level horizon, I don't know what might cause it, check if you can activate the attitude estimator from a NuttX shell manually on serial 4/5 port (You can have a serial connection to serial 4/5 port through a FTDI cable, and QGroundcontrol connection in the same time).

You can also check the output of the logcat (when booting), by keeping the FTDI connection intact, and issuing the "reboot" command. This might be helpful to determine what might be causing that error.

jmcwil, 2016-07-20:
@MHageH  Thanks for the response and the ideas.

I got it to calibrate!! I did as you said and checked to see if I could turn on the attitude estimator.  I thought that I had done this, but apparently I only did it for my old board.  I did 'attitude_estimator_q start' in the NuttX Shell, opened up QGroundControl, and the calibration for level works. Thanks for the help!

For some reason, the main LED is still flashing red.  The only error that I am getting "PREFLIGHT FAIL: NO BARO SENSOR 0".  This output also shows up in the logcat.  Is the baro sensor something that I am supposed to attach to Pixhawk?
Here is the output of the logcat.
```
WARN  [px4io] CRCs match
INFO  [dataman] Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
nsh: ms5611: command not found
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V2
WARN  [hmc5883] no device on bus 2
read_reg failWARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 1
ERROR [mpu6000] no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
nsh: meas_airspeed: command not found
nsh: ets_airspeed: command not found
nsh: ets_airspeed: command not found
nsh: sf10a: command not found
ERROR [sensors] no barometer found on /dev/baro0 (2)
WARN  [sensors] FATAL: No barometer found: /dev/baro0 (2)
ERROR [sensors] sensor initialization failed
ERROR [sensors] no barometer found on /dev/baro0 (2)
ERROR [sensors] no barometer found on /dev/baro0 (2)
INFO  [px4io] default PWM output device
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [ver] match: PX4FMU_V2
px4flow [165:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Fixedwing
INFO  [init] Mixer: /etc/mixers/Q.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1

NuttShell (NSH)
nsh> WARN  [mavlink] Disabling hardware flow control
WARN  [mavlink] Disabling hardware flow control
WARN  [commander] Not ready to fly: Sensors not set up correctly
```

Thank you!

MHageH, 2016-07-20:
@jmcwil you're welcome. 
By looking at the logcat, it seems that a lot of modules are disactivated, usually, you don't have to connect anything to the pixhawk except the red switch LED. Which firmware are you flashing, master or stable ? 
Try to flash the software provided by QGroundcontrol and see if it has the same effect, I don't think that there's hardware faults on this pixhawk too, since it's brand new.
Once it's all done, you need to tune some parameters too in QGroundControl to allow offboard control.
Check this [link](https://mhageh.gitbooks.io/c-uart-interface-technical-details/content/configuration.html), it contains the configuration necessary for this kind of control.

jmcwil, 2016-07-20:
@MHageH  Thanks for the input.  I am flashing my branch of master firmware.  I will try flashing the stable firmware directly from QGroundControl and see if that fixes anything.
I agree, this seems to be a software issue.  Thanks for pointing out the configurations for this!

jmcwil, 2016-07-20:
@MHageH Thanks again for your help, it turns out it was a problem with my firmware.

MHageH, 2016-07-20:
@jmcwil Glad it worked.

AKremor, 2017-01-05:
Hi all,

I appear to be running into the same problem as described in the first post. My setup is as follows:
- PixHawk PX4 running ArduCopter 3.4.3 I believe.
- USB Telemetry radio connected to TELEM 1 on the PX4.
- USB cable connected directly to the PX4's USB port on the side.
- It appears all sensors are set up correctly, as I am able to arm the quadcopter.

When going through the tutorial, I am having trouble completing step 1, and step 2.

**Problem: Step 1**
When I connect over the USB cable or telemetry link, I find that a large amount of 'gibberish' characters are displayed, and I am unable to get a nsh prompt up. I imagine these 'gibberish' characters could be the MAVLink messages? If I use Mission Planner or MAVProxy, I am able to get to the nsh prompt. I imagine this is sufficient, but given my setup isn't working, I am not sure.

**Problem: Step 2**
When I open up the nsh shell, and attempt to run `mavlink start -d /dev/ttyACM0`, I am presented with a `command not found` error. When listing all of the available apps in nsh, `mavlink` is not there. I am not too sure how to proceed past this problem.

Consequently, when I run the C program, I receive the SYS and COMP ID's correctly, but the program then stalls, waiting for the NED position to be sent I assume?
If anyone has any thoughts on this, I would greatly appreciate them.

Thank you in advance.

AKremor, 2017-01-05:
I seem to have resolved the above problem by loading a different firmware via QGroundControl.

Specifically, I have loaded the `PX4 Flight Stack Release v1.5.3` from the Firmware tab of QGroundControl. I did not choose the `ArduPilot Flight Stack` option.

This has made `mavlink` available from nsh, and resolved the `nsh: mavlink: command not found` error I was receiving.

closed 2018-05-19 by LorenzMeier