mavlink/c_uart_interface_example #18: Couldn't find <common/mavlink.h> file

LittleBun, 2015-11-26:
hi,
    When I run "make" to build this project, it shows the library included in mavlink_control.h <common/mavlink.h> couldn't be found. Any ideas?

LorenzMeier, 2015-11-26:
Run git submodule init and update as per the documentation

closed 2015-11-26 by LorenzMeier

LittleBun, 2015-12-01:
Sorry, but where I can find the documentation? I now can build the code, but after I login with "screen", I can't type on the shell to run the step 2: nsh> mavlink start -d  /dev/ttyACM0.
I can also run step 3, and the output stop at "CHECK FOR HEARTBEAT". 
Sorry for being silly, but I am new to MAVLink, could you help me out a little please?

JMare, 2015-12-05:
Hi mate, I have never been able to get a nsh shell on the usb port of my pixhawk, not sure why, but I can always get a shell through serial 5 and a ftdi board. It should be possible over usb of course, but you could try this if you have a ftdi lying around (make sure you are on serial 5 not 4)