mavlink/c_uart_interface_example #24: Switch custom mode and base mode

ptit-tom, 2017-05-03:
Hi everyone 

I'm working on offboard control of multicopter using c_uart_interface, pixhawk and mavlink. 
It might seem useless to know how to switch custom and base mode's autopilot for me but I'm curious and I don't understand the way to do it.

I saw https://dev.px4.io/en/concept/flight_modes.html , px4_custom_mode.h, MAV_MODE, MAV_MODE_FLAG, commander_state.msg in firmware but it muddled up to me the spirit !

What the difference between custom mode, base mode, main mode, main state and how use it with mavlink ?

Thanks you in advance, 
Thomas

MHageH, 2017-05-03:
In this [part](https://github.com/mavlink/c_uart_interface_example/blob/master/autopilot_interface.cpp#L529) code, you can see clearly how the interface sends the offboard command to the pixhawk.

I believe you can mod it the way you like to switch between modes.
Check the MAVLink [specification](https://pixhawk.ethz.ch/mavlink/) for further information.

On another note, in what circumstances are you trying to implement the Interface?

P.S : If I recall correctly, for the offboard mode, you should start to stream something first before activating it otherwise it will just turn off automatically on the pixhawk's end.

ptit-tom, 2017-05-05:
Thank you for your response MHageH ! 

I have already seen the code's part you showed me and it's not really what I mean. However, I don't have RC to control my MAV, so switch mode witch RC isn't a solution for me ! ahah

I'm in internship and my aim is to control MAV with companion computer for 3D mapping. 
I see in this website (https://dev.px4.io/en/concept/flight_modes.html) and in the file px4_custom_mode.h of PX4's firmware, modes don't correspond. For example, there is a sub_mode called AUTO_TAKEOFF apparently but I don't how switch to this mode. For the moment, I use MAV_CMD_NAV_TAKEOFF but my MAV doesn't move when I use it in HITL whereas a message appears :"takeoff detected"

Did you mean REQUEST_DATA_STREAM msg ? 

PS : I have read your report one month ago and your code. It's not really the same method but still useful.

MHageH, 2017-05-09:
I had a similar internship when I modified the interface (didn't have a RC neither).
Did you try to use SITL ? I've nearly made all of my flights in simulation first then tested on ground later on. 

As for the mods, I simply neglected them. I assumed that the drone will be controlled by the companion computer all the time, so I made a custom "takeoff" and hovering, with offboard mode on since the start. It's a bit dangerous this way, but I made sure that the drone will drop in safe mode if anything goes wrong. Unfortunately, I am no longer in position of the materials to test this for myself. 

Think about these points :
* Try to send the offboard command first then this MAV_CMD_NAV_TAKEOFF later on (before sending any setpoint).
* Are you sure that you respected all of MAV_CMD_NAV_TAKEOFF's params? What is weird is that it seems to detect the take off but doesn't do anything. Try another command 
otherwise.

If neither of these works, a simple [search](https://github.com/PX4/Firmware/search?utf8=%E2%9C%93&q=PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF&type=) indicates that PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF is passed over from mavlink_messages.cpp to commander.cpp (main app) and is treated at line 773 (case vehicle_command_s::VEHICLE_CMD_DO_SET_MODE).

I think you should check param2 and 3 of MAV_CMD_DO_SET_MODE and send it in. 

* REQUEST_DATA_STREAM is deprecated for now, and no it wasn't the thing I meant.

ptit-tom, 2017-05-10:
Hi MHageH !

Yes, I use it since 2 weeks ago and more exactly HITL. In fact, I don't really understand the difference between HITL and SITL.
Thanks, I do all of my commands in offboard mode now and it's work well, thanks to you !

About take off command, I have re-written my parameters : 
mavlink_command_long_t takeoff_cmd;
    /* Takeoff from ground / hand
     * |Minimum pitch (if airspeed sensor present), desired pitch without sensor
     * | Empty
     * | Empty
     * | Yaw angle (if magnetometer present), ignored without magnetometer
     * | Latitude
     * | Longitude
     * | Altitude|  */
    takeoff_cmd.command = MAV_CMD_NAV_TAKEOFF;
    takeoff_cmd.param1 = 0;
    takeoff_cmd.param4 = 0;
    takeoff_cmd.param5 = current_messages.gps_raw_int.lat;
    takeoff_cmd.param6 = current_messages.gps_raw_int.lon;
    takeoff_cmd.param7 = (current_messages.gps_raw_int.alt)/1000 + hauteur;
    takeoff_cmd.target_system = system_id;
    takeoff_cmd.confirmation = true;

For PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF, I 've already tried to use  MAV_CMD_DO_SET_MODE but I didn't find which rights param1 and param2 to set for change in mode I wish. In px4_custom_mode.h , only one MAIN_MODE and one SUB_MODE get values... Do you think I have to set values for others modes to use it ?

MHageH, 2017-05-10:
Hello again,
Good thing it works well now. 
As for [HITL](https://dev.px4.io/en/simulation/hitl.html), it is basically a simulation where you use the real hardware (pixhawk) to fly the drone in simulation (with qgroundcontrol), whereas in [SITL](https://dev.px4.io/en/simulation/sitl.html) it is solely based on software, where you can see a [virtual drone flying](http://imgur.com/a/PAsTr). 

Try to specify param2 as PX4_CUSTOM_MAIN_MODE_AUTO and param3 as PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF with the MAV_CMD_DO_SET_MODE command as described with the [PX4 Custom modes](https://github.com/PX4/Firmware/blob/master/src/modules/commander/px4_custom_mode.h).
As param1, send MAV_MODE_AUTO_ARMED (described in MAV_MODE enum in MAVlink reference).

I think the sub_modes are only defined for the AUTO_MOD values. Try to send only the PX4_CUSTOM_MAIN_MODE for other modes (not the sub ones), and see if it works.

Your approach is rather interesting. I can implement this later on if you can test it with my modification of the interface to rather command the drone in proper mods, rather than all the time in offboard mode.

ptit-tom, 2017-05-11:
Ok, thanks for the information. I use HITL but I check the result with jmavsim and not with qgroundcontrol, actually.

I 've checked for MAV_CMD_DO_SET_MODE and I 've seen little "problem". 
In [c_uart_interface directory](https://github.com/mavlink/c_library_v1/blob/7d0aa0205ef8bd13bf855b21d3ebe054dc9109ef/common/common.h), the command got 2 params.
In [mavlink website](https://pixhawk.ethz.ch/mavlink/), this same command got 3 params...

I will try with main mode only like you said me. I shall come back to you when it's done.
Yes, if you want I can test it for you with my material. But for the moment, I only got Pixhawk and jmavsim.

MHageH, 2017-05-11:
I actually went through this when I was developing my version of the interface. 
C_uart_interface isn't compatible with the latest MAVLink library (was testing it on jmavsim and the real drone back then), that's why I downloaded that [version](https://github.com/mavlink/c_library_v1/tree/7d0aa0205ef8bd13bf855b21d3ebe054dc9109ef)
(commit: 7d0aa0205ef8bd13bf855b21d3ebe054dc9109ef) and fixed it in my mod.

You are rather right, I didn't notice that at first.
If you look at the MAVLink version used 3 years ago (actually used as [a submodule](https://github.com/mavlink/c_uart_interface_example/tree/master/mavlink/include/mavlink) in C_UART_Interface), you'll see that [MAV_CMD_DO_SET_MODE](https://github.com/mavlink/c_library_v1/blob/7d0aa0205ef8bd13bf855b21d3ebe054dc9109ef/common/common.h#L341) doesn't have a CUSTOM_SUB_MODE, whereas for the [current version](https://github.com/mavlink/c_library_v1/blob/429bc35793ededd26be7ed3484e70e14416d1ef4/common/common.h#L389), it is defined.
I think that the [MAVLink reference](https://pixhawk.ethz.ch/mavlink/) is automatically generated with the latest compilation of [MAVLink](https://github.com/mavlink/c_library_v1) library.

It is actually very similar with [SET_MODE](https://pixhawk.ethz.ch/mavlink/#SET_MODE) command before it was deprecated. 

I'm afraid that we are not capable to set custom submods with this version. Current PX4 [firmware](https://github.com/PX4/Firmware/blob/master/src/modules/commander/px4_custom_mode.h) do understand it, but there's no way to send it.

Perhaps the only solution would be MAV_CMD_NAV_TAKEOFF as you've done before. 

Update me with your latest findings.

ptit-tom, 2017-06-19:
Hi MHageH ! 

I've found something interesting about offboard's submode. Look at [this](http://discuss.px4.io/t/px4-hidden-flight-bitmasks/1371), there are hidden mode linked with 3 mavlink's message : set_position_target_local_ned, set_attitude_target and set_actuator_control_target 
For more details, look at [mavlink_receiver.cpp](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp).

About PX4_CUSTOM  mode, It did'nt work so I've give it up.

closed 2018-05-19 by LorenzMeier