mavlink/c_uart_interface_example #25: It stopped at he lline "GOT AUTOPILOT COMPONENT ID: 1".

goodman110110110, 2018-03-16:
hello
I make the c_uart_interface_example and use the command ./mavlink_control -d /dev/ttyACM0.
but the programme cannot go on. It stopped at he lline "GOT AUTOPILOT COMPONENT ID: 1".
Please help  me. 
Thank you.
what have I done as follows:
__________________________________________________
ghf@ghf-A8SR:~$ cd c_uart_interface_example/
ghf@ghf-A8SR:~/c_uart_interface_example$ ./mavlink_control -d /dev/ttyACM0
OPEN PORT
Connected to /dev/ttyACM0 with 57600 baud, 8 data bits, no parity, 1 stop bit (8N1)

START READ THREAD 

CHECK FOR MESSAGES
Found

GOT VEHICLE SYSTEM ID: 1
GOT AUTOPILOT COMPONENT ID: 1

MHageH, 2018-03-16:
Usually it's a problem with the LOCAL_POSITION_NED which the Pixhawk isn't sending. This was discussed before, reference is [here](https://github.com/mavlink/c_uart_interface_example/issues/16).

goodman110110110, 2018-03-17:
Tank you, I checked and succeed.

closed 2018-03-17 by goodman110110110