mavlink/c_uart_interface_example #26: Not working

grcasanova, 2018-05-19:
Tried with Pixhawk 2.4.8 and the latest ardupilot firmware, no messages are retrieved, the implementation is supposedly broken as the Mission planner can correctly get messages.

LorenzMeier, 2018-05-19:
This is a MAVLink example, not an APM example. There is special magic required there to enable telemetry, so nothing is broken here.

closed 2018-05-19 by LorenzMeier

hamishwillee, 2018-05-21:
@grcasanova Messages **are** actually received on Pixhawk (Pixracer) with latest ArduPilot firmware, as kindly tested by @PeterBarker. The problem appears to be that we only print out two messages LOCAL_POSITION_NED, HIGHRES_IMU - the first will only be printed out if you have a position lock and the second is not emitted by ArduPilot. 

We're catching a number of other messages here: https://github.com/mavlink/c_uart_interface_example/blob/master/autopilot_interface.cpp#L265
and these could be printed out here: https://github.com/mavlink/c_uart_interface_example/blob/master/mavlink_control.cpp#L247

Note that I am not sure that the example would correctly enter and stay in offboard mode on ArduPilot either. 

This is still a useful point for developing the C++ interface on any mavlink system. This is not a bug in the example, but could certainly be improved so that it behaves meaningfully on ArduPilot as well as PX4.

If you can create a PR to show some messages that are also exported by Ardupilot  - e.g. heartbeat - then we'd accept it.