mavlink/c_uart_interface_example #28: Can`t  make

ma18042, 2018-06-25:
Hello.

when I make mavlink/c_uart_interface_example ,error occoured.
I want to know how solve this problem.
like this ↓

ubuntu@ubuntu-CFSZ5-3:~/c_uart_interface_example$ make
git submodule update --init --recursive
Submodule 'mavlink/include/mavlink/v1.0' (git://github.com/mavlink/c_library.git) registered for path 'mavlink/include/mavlink/v1.0'
Cloning into 'mavlink/include/mavlink/v1.0'...
fatal: unable to connect to github.com:
github.com[0: 192.30.255.113]: errno=Connection timed out
github.com[1: 192.30.255.112]: errno=Connection timed out

fatal: clone of 'git://github.com/mavlink/c_library.git' into submodule path 'mavlink/include/mavlink/v1.0' failed
makefile:7: recipe for target 'git_submodule' failed
make: *** [git_submodule] Error 128

ma18042, 2018-06-25:
sorry,I solved.

closed 2018-06-25 by ma18042

suraj2596, 2019-11-27:
Im having the same issue. May I know how you solved it? Thanks :)

suraj2596, 2019-11-27:
So, this helped.
`git config --global url."https://".insteadOf git://`
Thanks to https://stackoverflow.com/questions/16298986/unable-to-connect-to-github-com-for-cloning