mavlink/c_uart_interface_example #33: `current_setpoint` should be guarded with a mutex

allsey87, 2020-07-30:
I have been going through this code and thought I would point out that `current_setpoint` is written from the main thread and read from the _writing thread_ without any protection (i.e., a mutex or a lock). Although unlikely, it possible that `current_setpoint` is read by the _writing thread_ while being written from the main thread which could result in things going to hell very quickly since we are talking about a quadcopter or similar here.

In contrast to this, I am almost certain that reading and writing to the serial port from different threads just works in POSIX and doesn't need any mutexes.

hamishwillee, 2020-07-31:
@allsey87 We'd welcome a PR! @julianoes FYI

allsey87, 2020-07-31:
Well, I have used this as a starting point to write my own MAVLink interface for integration into the [ARGoS Simulator](argos-sim.info/). During this process, I have made a far more simplified version of this example which I can share with you if you are interested

hamishwillee, 2020-08-03:
@allsey87 Yes, would appreciate a PR of new/updated example. I'm not particularly good at C/C++ but I'll ask another team member to look at it.

hamishwillee, 2020-08-04:
Fixed. Closing :-). Thanks! Still interested in a PR if you think it better highlights how to approach this problem.

PR #34 (2020-08-04): Add mutex around current_setpoint
Closes #33.
commit d375d331 (2020-08-04): Add mutex around current_setpoint

Closes #33.
closed 2020-08-04 by hamishwillee