mavlink/c_uart_interface_example #36: can not simulate with jmavsim or Gazebo

OneOfManyyy, 2021-02-24:
Hello, new user here. I tried to simulate the example application with both Jmavsim and Gazebo by running "./mavlink_control -u 127.0.0.1 -a". I see on the QGroundControl that the drone gets armed at the start and disarmed in the end so I think the commands are sent successfully but there is no takeoff, no movement and no landing happens. What am I doing wrong?

hamishwillee, 2021-02-24:
No idea. What do you see on the console? https://github.com/mavlink/c_uart_interface_example#execution

This is pretty low level code. What are you ultimately trying to do? If it is "control a drone" then you would be better off looking at the MAVSDK https://mavsdk.mavlink.io/develop/en/index.html

OneOfManyyy, 2021-02-27:
Thanks a lot for your response. I was able to run the code by commenting one of the "if" statements which I suppose was disabling the position to be changed. I am trying to develop algorithms for multiple drone movements but since I am using a microcontroller with FreeRTOS on top, as far as I understood MAVSDK can not be a choice.

hamishwillee, 2021-02-28:
>  as far as I understood MAVSDK can not be a choice.

That is a question to ask on the PX4 slack channel #mavsdk. It may be that FreeRTOS is not a supported platform, but would still work if you built from source.

xiaoshengjianan, 2023-12-26:
> commenting one of the "if" statements

Could you reply in detail?

julianoes, 2023-12-26:
> It may be that FreeRTOS is not a supported platform, but would still work if you built from source.

MAVSDK requires threads, networking, and serial APIs as provided by Linux/Unix and Windows. There is a chance that it would work on top of NuttX which is mostly POSIX compliant, but likely not FreeRTOS.

So this means you will have to learn how to mess with MAVLink at the lower level, using something like this repo as a starting point. This will likely be difficult to get going, so testing as much as possible on the simulator makes sense.

Now, when it comes to certain commands and why they don't work, one way would be to check what MAVSDK is doing and copy that.