mavlink/c_uart_interface_example #37: Can not fly !

1556260456, 2021-07-17:
Hello, I am trying to fly my quadrocopter drones. And my platform is NVIDIA Xavier-(Linux)，I use this uart-interface and the example in it ,try to let the uav fly to 2m. I had the blades removed indoors for the experiment, and the GPS signal was good，after I run mavlink_control , I can see the flight mode changed into offboard,But none of the four motors responded, and soon the safety switch was locked again. How conld I make the uav fly to 2m by sending mavlink message from linux uart?

hamishwillee, 2021-07-18:
What flight stack? Are you streaming the offboard commands? If so, at what rate?

1556260456, 2021-07-20:
That is my mistake, I did not notice this code,after I ignore them,everything is OK
`if(autotakeoff)
{
    sp.type_mask  |=MAVLINK_MSG_SET_POSITION_TARGET_LOCAL_NED_TAKEOFF
}
`

closed 2021-07-20 by 1556260456