mavlink/c_uart_interface_example #39: It stopped at "CHECK FOR MESSAGES"

ldkcumt, 2022-12-05:
.Using a UART (serial) connection
.TELEM2
.The USB-to-serial adapter board is not a FTDI chip, but a CH340G chip of China.
.Then using a USB connection, it also stopped at here.

hamishwillee, 2022-12-07:
It is impossible to debug this for you. However the first thing I'd be doing is making sure that the thing you are connecting to is actually sending any messages. Consider using Wireshark to see what is on the interface.

You could also connect MAVSDK or QGroundControl and confirm they are seeing messages.

closed 2022-12-07 by hamishwillee

TarekElDick, 2023-04-24:
It's a common mistake to connect the Tx pin to Tx and Rx pin to Rx when wiring devices for serial communication. Always remember to connect the Tx pin of one device to the Rx pin of the other device and vice versa. If that doesnt work here are some other things to consider.

    Check the wiring: Double-check the connections between the Pixhawk and the companion computer. Ensure that the Tx, Rx, and GND pins on the telemetry port (TELEM2) are properly connected to the FTDI cable. The Pixhawk's Tx pin should be connected to the FTDI cable's Rx pin, and the Pixhawk's Rx pin should be connected to the FTDI cable's Tx pin.

    Verify the FTDI cable: Make sure the FTDI cable is working correctly. You can test it on another device or with a loopback test by shorting the Tx and Rx pins on the FTDI cable and using a serial terminal to send and receive data.

    Configure the Pixhawk: Ensure that the Pixhawk's telemetry port parameters are correctly set in the ground control station (GCS) software (e.g., SER_TEL2_BAUD or TELEM2_BAUD). The baud rate should match the one you are using on the companion computer.

    Configure the companion computer: Check the software on the companion computer to ensure that the correct port (e.g., /dev/ttyUSB0), baud rate, and MAVLink protocol version are set. Ensure that the appropriate FTDI drivers are installed on the companion computer.

    Power cycle the Pixhawk and companion computer: Turn off both the Pixhawk and the companion computer, then power them back on. This can help resolve any temporary communication issues.

    Update Pixhawk firmware: Make sure your Pixhawk is running the latest firmware version, as this can improve compatibility with the companion computer and MAVLink communication.

    Check for hardware issues: Inspect the Pixhawk and the companion computer for any visible signs of damage or loose connections.

    Monitor communication: Use a software tool like Wireshark, MAVProxy, or a serial terminal to monitor the communication between the Pixhawk and the companion computer. This can help you identify any issues or errors in the MAVLink messages.