mavlink/c_uart_interface_example #41: Timestamps reference

DanieleMarchisotti, 2023-05-05:
In ardupilot_interface.cpp the timestamps of MavLink messages can be obtained with "current_messages.time_stamps".
Which is the reference of these timestamps?
Are they directly referred to GPS timestamp or they are referred to the internal clock of the autopilot? I refer to outdoor conditions with optimal GPS signal.

julianoes, 2023-05-05:
Can you give some context of the problem you're having? I don't quite understand the technicalities otherwise.

DanieleMarchisotti, 2023-05-09:
I need to the autopilot data labeled with GPS timestamps to have data aligned with another system that has data referred to GPS timestamp. Anyway, I was able to solve the problem by reading the SYSTEM_TIME message.

closed 2025-06-25 by DanieleMarchisotti